LennardKittner/HyperHeadset #1: Fedora 38 Broken pipe issue

tottoszsolt, 2023-06-15:
I have installed the dependencies, but when I try to launch it, I get broken pipes error, both the cli and the hyperx app, please help. Thank you!

LennardKittner, 2023-06-15:
Ok, strange, I never had this error before.
What version of the headset are you using (HP or HyperX)?
You can run `lsusb` and then look for the string "HyperX Cloud II Wireless" to find the vendor name.
Where does the error occur could you send me the full terminal output?

tottoszsolt, 2023-06-15:
lsusb:
Bus 003 Device 004: ID 0951:1718 Kingston Technology HyperX Cloud II Wireless
terminal output:
[image]

LennardKittner, 2023-06-15:
Ok, you have the older version of the headset.
I have the newer version and hoped that only the vendorID and productID would be different between the versions.
Have you ever tried running the app without sudo?
Have you compiled the app your self or downloaded the pre-compiled version?

tottoszsolt, 2023-06-15:
I used the precompiled version, I will check the self compiled in the morning

LennardKittner, 2023-06-15:
Ok, if you do, could you replace the following lines in `src/lib.rs"? This will make the app crash immediately and may tell me more about where and why the error happens.
81: `let hid_api = HidApi::new().unwrap();`
88: `}).unwrap().unwrap();`
125: `self.hid_device.write(&BATTERY_PACKET).unwrap();`
127: `let res = self.hid_device.read_timeout(&mut buf[..], 1000).unwrap();`
I'm sorry for your inconvenience :(.

LennardKittner, 2023-06-16:
I have compiled it for you with the changes I mentioned above.
[cli_app.zip](https://github.com/LennardKittner/HyperXCloudIIWireless/files/11766748/cli_app.zip)

tottoszsolt, 2023-06-16:
Thank you! I will try it in a moment!

tottoszsolt, 2023-06-16:
[image]
This is the error I get trying it to run withouth sudo,
and this is what I get running with sudo 
[image]

LennardKittner, 2023-06-16:
The first error indicates that the rules are not set correctly, so the app does not have the necessary permissions to access the headset (but we can fix this later).

The second error indicates that the packet I used to request the battery level failed. The reason for this could be that your headset (the older version) expects different packets than mine.
I have seen another project on GitHub that uses the older version. I will try to use the packet from that repository and I will send you another executable in a couple of hours. I have to go to the university now.

tottoszsolt, 2023-06-16:
Okay, thank you!

LennardKittner, 2023-06-16:
Could you please run these two executables with sudo and let me know what happens? [cli_app.zip](https://github.com/LennardKittner/HyperXCloudIIWireless/files/11770182/cli_app.zip)
I have changed the packet that is sent to the headset and hope that this will fix the issue.

tottoszsolt, 2023-06-16:
Sorry for the late reply, here are the results
[image]

LennardKittner, 2023-06-16:
Could you please try this again [cli_app.zip](https://github.com/LennardKittner/HyperXCloudIIWireless/files/11771577/cli_app.1.zip).
If that does not work either, try this: https://github.com/Agustin-Jerusalinsky/hyperx-cloud-II-battery/.
I think the problem is that the older version of the headset expects different packets.
Therefore, if both apps do not work, I think you will have to use Wireshark on Windows so we can figure out what the right packets are.

LennardKittner, 2023-06-16:
One more alternative would be that you try to compile the app yourself.

tottoszsolt, 2023-06-16:
Its not working anyhow sadly.. I am trying to compile it myself, but I am very noob in the Linux field, so I tried avoiding it tbh

tottoszsolt, 2023-06-16:
While trying to compile it myself, I get this errors: 

totto@zsolti-pc HyperXCloudIIWireless-main]$ cargo build --release
    Updating crates.io index
   Compiling pkg-config v0.3.27
   Compiling libc v0.2.146
   Compiling proc-macro2 v1.0.60
   Compiling quote v1.0.28
   Compiling unicode-ident v1.0.9
   Compiling unicode-width v0.1.10
   Compiling strsim v0.8.0
   Compiling vec_map v0.8.2
   Compiling bitflags v1.3.2
   Compiling ansi_term v0.12.1
   Compiling xml-rs v0.8.14
   Compiling cc v1.0.79
   Compiling thiserror v1.0.40
   Compiling textwrap v0.11.0
   Compiling libdbus-sys v0.2.5
error: failed to run custom build command for `libdbus-sys v0.2.5`

Caused by:
  process didn't exit successfully: `/home/totto/Downloads/HyperXCloudIIWireless-main/target/release/build/libdbus-sys-9400c167139cd70f/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-changed=build.rs
  cargo:rerun-if-changed=build_vendored.rs
  cargo:rerun-if-env-changed=DBUS_1_NO_PKG_CONFIG
  cargo:rerun-if-env-changed=PKG_CONFIG_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG
  cargo:rerun-if-env-changed=PKG_CONFIG
  cargo:rerun-if-env-changed=DBUS_1_STATIC
  cargo:rerun-if-env-changed=DBUS_1_DYNAMIC
  cargo:rerun-if-env-changed=PKG_CONFIG_ALL_STATIC
  cargo:rerun-if-env-changed=PKG_CONFIG_ALL_DYNAMIC
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_PATH
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_LIBDIR
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_SYSROOT_DIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR

  --- stderr
  pkg_config failed: `PKG_CONFIG_ALLOW_SYSTEM_CFLAGS="1" PKG_CONFIG_ALLOW_SYSTEM_LIBS="1" "pkg-config" "--libs" "--cflags" "dbus-1" "dbus-1 >= 1.6"` did not exit successfully: exit status: 1
  error: could not find system library 'dbus-1' required by the 'libdbus-sys' crate

  --- stderr
  Package dbus-1 was not found in the pkg-config search path.
  Perhaps you should add the directory containing `dbus-1.pc'
  to the PKG_CONFIG_PATH environment variable
  Package 'dbus-1', required by 'virtual:world', not found
  Package 'dbus-1', required by 'virtual:world', not found

  One possible solution is to check whether packages
  'libdbus-1-dev' and 'pkg-config' are installed:
  On Ubuntu:
  sudo apt install libdbus-1-dev pkg-config
  On Fedora:
  sudo dnf install dbus-devel pkgconf-pkg-config

  thread 'main' panicked at 'explicit panic', /home/totto/.cargo/registry/src/index.crates.io-6f17d22bba15001f/libdbus-sys-0.2.5/build.rs:25:9
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
warning: build failed, waiting for other jobs to finish...
error: failed to run custom build command for `libdbus-sys v0.2.5`

Caused by:
  process didn't exit successfully: `/home/totto/Downloads/HyperXCloudIIWireless-main/target/release/build/libdbus-sys-9400c167139cd70f/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-changed=build.rs
  cargo:rerun-if-changed=build_vendored.rs
  cargo:rerun-if-env-changed=DBUS_1_NO_PKG_CONFIG
  cargo:rerun-if-env-changed=PKG_CONFIG_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG
  cargo:rerun-if-env-changed=PKG_CONFIG
  cargo:rerun-if-env-changed=DBUS_1_STATIC
  cargo:rerun-if-env-changed=DBUS_1_DYNAMIC
  cargo:rerun-if-env-changed=PKG_CONFIG_ALL_STATIC
  cargo:rerun-if-env-changed=PKG_CONFIG_ALL_DYNAMIC
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_PATH
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_LIBDIR
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_SYSROOT_DIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR

  --- stderr
  pkg_config failed: `PKG_CONFIG_ALLOW_SYSTEM_CFLAGS="1" PKG_CONFIG_ALLOW_SYSTEM_LIBS="1" "pkg-config" "--libs" "--cflags" "dbus-1" "dbus-1 >= 1.6"` did not exit successfully: exit status: 1
  error: could not find system library 'dbus-1' required by the 'libdbus-sys' crate

  --- stderr
  Package dbus-1 was not found in the pkg-config search path.
  Perhaps you should add the directory containing `dbus-1.pc'
  to the PKG_CONFIG_PATH environment variable
  Package 'dbus-1', required by 'virtual:world', not found
  Package 'dbus-1', required by 'virtual:world', not found

  One possible solution is to check whether packages
  'libdbus-1-dev' and 'pkg-config' are installed:
  On Ubuntu:
  sudo apt install libdbus-1-dev pkg-config
  On Fedora:
  sudo dnf install dbus-devel pkgconf-pkg-config

  thread 'main' panicked at 'explicit panic', /home/totto/.cargo/registry/src/index.crates.io-6f17d22bba15001f/libdbus-sys-0.2.5/build.rs:25:9
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
[totto@zsolti-pc HyperXCloudIIWireless-main]$

LennardKittner, 2023-06-16:
It complains about dbus not being installed, but dbus is only used for the tray application.
try to run `cargo build --bin cli_app`.
If this does not work either remove the two last lines from `Cargo.toml` and then run `cargo build --bin cli_app` again.

tottoszsolt, 2023-06-16:
compiling with cargo build --bin cli_app: 
    Updating crates.io index
   Compiling pkg-config v0.3.27
   Compiling libc v0.2.146
   Compiling proc-macro2 v1.0.60
   Compiling unicode-ident v1.0.9
   Compiling quote v1.0.28
   Compiling unicode-width v0.1.10
   Compiling vec_map v0.8.2
   Compiling strsim v0.8.0
   Compiling bitflags v1.3.2
   Compiling ansi_term v0.12.1
   Compiling xml-rs v0.8.14
   Compiling thiserror v1.0.40
   Compiling cc v1.0.79
   Compiling textwrap v0.11.0
   Compiling libdbus-sys v0.2.5
error: failed to run custom build command for `libdbus-sys v0.2.5`

Caused by:
  process didn't exit successfully: `/home/totto/Downloads/HyperXCloudIIWireless-main/target/debug/build/libdbus-sys-7db97899a5e261a6/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-changed=build.rs
  cargo:rerun-if-changed=build_vendored.rs
  cargo:rerun-if-env-changed=DBUS_1_NO_PKG_CONFIG
  cargo:rerun-if-env-changed=PKG_CONFIG_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG
  cargo:rerun-if-env-changed=PKG_CONFIG
  cargo:rerun-if-env-changed=DBUS_1_STATIC
  cargo:rerun-if-env-changed=DBUS_1_DYNAMIC
  cargo:rerun-if-env-changed=PKG_CONFIG_ALL_STATIC
  cargo:rerun-if-env-changed=PKG_CONFIG_ALL_DYNAMIC
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_PATH
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_LIBDIR
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_SYSROOT_DIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR

  --- stderr
  pkg_config failed: `PKG_CONFIG_ALLOW_SYSTEM_CFLAGS="1" PKG_CONFIG_ALLOW_SYSTEM_LIBS="1" "pkg-config" "--libs" "--cflags" "dbus-1" "dbus-1 >= 1.6"` did not exit successfully: exit status: 1
  error: could not find system library 'dbus-1' required by the 'libdbus-sys' crate

  --- stderr
  Package dbus-1 was not found in the pkg-config search path.
  Perhaps you should add the directory containing `dbus-1.pc'
  to the PKG_CONFIG_PATH environment variable
  Package 'dbus-1', required by 'virtual:world', not found
  Package 'dbus-1', required by 'virtual:world', not found

  One possible solution is to check whether packages
  'libdbus-1-dev' and 'pkg-config' are installed:
  On Ubuntu:
  sudo apt install libdbus-1-dev pkg-config
  On Fedora:
  sudo dnf install dbus-devel pkgconf-pkg-config

  thread 'main' panicked at 'explicit panic', /home/totto/.cargo/registry/src/index.crates.io-6f17d22bba15001f/libdbus-sys-0.2.5/build.rs:25:9
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
warning: build failed, waiting for other jobs to finish...

tottoszsolt, 2023-06-16:
Tried compiling after installing the missing dependencies: 
   Compiling pkg-config v0.3.27
   Compiling libc v0.2.146
   Compiling proc-macro2 v1.0.60
   Compiling unicode-width v0.1.10
   Compiling unicode-ident v1.0.9
   Compiling quote v1.0.28
   Compiling vec_map v0.8.2
   Compiling strsim v0.8.0
   Compiling bitflags v1.3.2
   Compiling ansi_term v0.12.1
   Compiling xml-rs v0.8.14
   Compiling thiserror v1.0.40
   Compiling cc v1.0.79
   Compiling textwrap v0.11.0
   Compiling libdbus-sys v0.2.5
   Compiling hidapi v2.3.3
   Compiling syn v2.0.18
   Compiling atty v0.2.14
   Compiling dbus v0.9.7
   Compiling clap v2.34.0
error: failed to run custom build command for `hidapi v2.3.3`

Caused by:
  process didn't exit successfully: `/home/totto/Downloads/HyperXCloudIIWireless-main/target/release/build/hidapi-5a158c9dd26ab897/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-changed=etc/hidapi/linux/hid.c
  cargo:rerun-if-env-changed=LIBUDEV_NO_PKG_CONFIG
  cargo:rerun-if-env-changed=PKG_CONFIG_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG
  cargo:rerun-if-env-changed=PKG_CONFIG
  cargo:rerun-if-env-changed=LIBUDEV_STATIC
  cargo:rerun-if-env-changed=LIBUDEV_DYNAMIC
  cargo:rerun-if-env-changed=PKG_CONFIG_ALL_STATIC
  cargo:rerun-if-env-changed=PKG_CONFIG_ALL_DYNAMIC
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_PATH
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_LIBDIR
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_SYSROOT_DIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR

  --- stderr
  thread 'main' panicked at 'Unable to find libudev: `PKG_CONFIG_ALLOW_SYSTEM_CFLAGS="1" PKG_CONFIG_ALLOW_SYSTEM_LIBS="1" "pkg-config" "--libs" "--cflags" "libudev"` did not exit successfully: exit status: 1
  error: could not find system library 'libudev' required by the 'hidapi' crate

  --- stderr
  Package libudev was not found in the pkg-config search path.
  Perhaps you should add the directory containing `libudev.pc'
  to the PKG_CONFIG_PATH environment variable
  Package 'libudev', required by 'virtual:world', not found
  ', /home/totto/.cargo/registry/src/index.crates.io-6f17d22bba15001f/hidapi-2.3.3/build.rs:58:54
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
warning: build failed, waiting for other jobs to finish...

LennardKittner, 2023-06-16:
You have to install hidapi `sudo dnf install hidapi`.

tottoszsolt, 2023-06-16:
I have already installed it

LennardKittner, 2023-06-16:
ah my bad it says `libudev` is missing.
`sudo dnf install rust-libudev-sys-devel`

tottoszsolt, 2023-06-16:
Damn.. I saw that too but cant find the install command for that.. cheking if its working..

LennardKittner, 2023-06-16:
I also found this but don't know what the difference is `sudo dnf install rust-libudev+default-devel`

tottoszsolt, 2023-06-16:
I have run it, but still, broken pipe error

tottoszsolt, 2023-06-16:
also, finally compiled with no error

LennardKittner, 2023-06-16:
Try `git checkout 1-fedora-38-broken-pipe-issue ` and build again.

tottoszsolt, 2023-06-16:
There is no git repository it says

LennardKittner, 2023-06-16:
are you in the root of the repo?

tottoszsolt, 2023-06-16:
Yeah, I was, now I see there is a separate repo.. I download that, and check in a moment

LennardKittner, 2023-06-16:
1-fedora-38-broken-pipe-issue  is just a branch of the same repo you are already in.

LennardKittner, 2023-06-16:
However, I suspect that this will still result in the same error because it seems like your headset version is incompatible with the packets I'm trying to send to it.
Therefore, the only option left is Wireshark.
I have a little tutorial in my README but if you have any questions, please ask.

tottoszsolt, 2023-06-17:
Is a Virtual machine good for it too? I dont have windows installed, only fedora

LennardKittner, 2023-06-17:
I have never tried it inside a VM, but I think it should work if you enable USB passthrough.

LennardKittner, 2023-06-19:
While working on another project to check the battery level of my mouse, I stumbled across the fact that a USB device can have multiple interfaces and that it can make a difference to which interface the packets are sent. On my headset, this makes no difference, but maybe on yours. I have created a new program to test all possible packets on all interfaces. If you want, you can try that.
I have included the program ([device_tester.zip](https://github.com/LennardKittner/HyperXCloudIIWireless/files/11790103/device_tester.zip)) but you can also compile it yourself if you want.
To do that:
1. checkout the issue branch
2. ` cargo build --bin device_tester `
3.  `sudo ./target/debug/device_tester` 

I hope we can resolve this issue.

tottoszsolt, 2023-06-19:
Looking into it in a sec

tottoszsolt, 2023-06-19:
Testing device: 2385:5912:3
packet: 0
hidapi error: Broken pipe
[]
packet: 1
hidapi error: Broken pipe
[]
packet: 2
hidapi error: Broken pipe
[]
packet: 3
hidapi error: Broken pipe
[]
Testing device: 2385:5912:3
packet: 0
hidapi error: Broken pipe
[]
packet: 1
hidapi error: Broken pipe
[]
packet: 2
hidapi error: Broken pipe
[]
packet: 3
hidapi error: Broken pipe
[]
Testing device: 2385:5912:3
packet: 0
hidapi error: Broken pipe
[]
packet: 1
hidapi error: Broken pipe
[]
packet: 2
hidapi error: Broken pipe
[]
packet: 3
hidapi error: Broken pipe
[]
Its just broken.. I dont get it...

LennardKittner, 2023-06-19:
Ok was worth a shot, then I think Wireshark is the only option left.

tottoszsolt, 2023-08-18:
I can make the WIreShark now.. I have a windows laptop I can do it on

tottoszsolt, 2023-08-18:
Also, do you have discord or something I can talk with you about, outside of Github?

miikeyjay, 2023-08-24:
hi, i'm having the same issue with the older cloud ii, is there any update on the support for it?

LennardKittner, 2023-08-24:
@tottoszsolt and I tried to find out what the Hyperx app sends to the old headset, but we were not successful.
When I use Wire Shark as described in the README, I find the package, which I have to send to the device
[image]
[0x06, 0xff, 0xbb, 0x02].

However, we were not able to locate a similar packet on his computer.
Thus, there are currently two possibilities:
1. Someone else tries to find the packet using Wire Shark and sends me the data I need.
2. I (or someone else) tries to reverse engineer the Hyperx app. I heard that it has no obfuscation and is pretty easy to disassemble with dnSpy. However, I already tried to disassemble it once and failed, but I also didn't have much time and am unfamiliar with dnSpy.

LennardKittner, 2023-08-24:
OK, If you have any questions please ask.

miikeyjay, 2023-08-24:
can you make a example of the packet you need, i used the print you sent as a example and got this
[image]

miikeyjay, 2023-08-24:
all responses are these
[image]

LennardKittner, 2023-08-24:
nice is your headset at 89%?

miikeyjay, 2023-08-24:
yes, it is at 89%

miikeyjay, 2023-08-24:
running the cli_app works now but the hyperx_x_cloud_ii_wireless doesn't
[image]

LennardKittner, 2023-08-24:
Strange the two apps use the same code to interact with the device.
I will look into it.

LennardKittner, 2023-08-29:
Yes, both applications use the same backend code; the only difference is that the cli application sends the battery request just once.
Does the tray application crash immediately, or does it show the battery level and then crash at some point?
If you execute the cli app multiple times, do you get the correct response every time?

miikeyjay, 2023-08-30:
when i run the tray app it waits a second and give the error message,
[image]
the cli app doesnt work every time, since the last 5 days it worked around 2 times only.
[image]

LennardKittner, 2023-08-30:
Interesting @tottoszsolt reports similar behavior. It seems like the packet sent to the headset only works once per power cycle.
I think this is because the headset is waiting for a specific answer but does not get one and thus shuts down the conversation.
Could you maybe look again with Wire Shark to see if you find any additional packets that are regularly sent between the host and headset, or vice versa?
You could also send me a dump of the traffic, and I'll take a look.

miikeyjay, 2023-09-23:
sorry for the delay, life happened, i tried to dump all the packets from the headset, if you need any other info on it feel free to ask.
[only cloud ii.zip](https://github.com/LennardKittner/HyperXCloudIIWireless/files/12707439/only.cloud.ii.zip)

LennardKittner, 2023-09-24:
Thank you, I will look into it :)

LennardKittner, 2023-10-27:
Hi, sorry for the late response. I checked the log, and the packet I sent matches the packet sent in the log. However, it seems like the host also sends some interrupts to the device; maybe these are the key? I don't see an option to send interrupts with the HID library I used in the project. I will try to find another HID library or to decompile the Windows app and look into what actually happens. Decompiling the Windows app is a bit annoying because, as far as I can tell, it is only available through the Windows Store.

miikeyjay, 2023-10-29:
No problem, if you need anything that i can prove to help just ping and i will try.

LennardKittner, 2023-10-29:
Hey, thank you for the offer. If you want you could look into how to disassemble the hyperx windows app or haw to get the hyperx windows app without using the windows store (if that is even possible).

miikeyjay, 2024-01-10:
after a while i started to search for information on this and found this python app that works with my headset (https://github.com/Agustin-Jerusalinsky/hyperx-cloud-II-battery) the [battery.py](http://battery.py/) returns the right %

LennardKittner, 2024-01-10:
Yes, I would expect that. The first packet that was tested here, https://github.com/LennardKittner/HyperXCloudIIWireless/issues/1#issuecomment-1692048386 is actually from that repository. However, I just noticed it does one thing different before the request for the battery level, it also sends a report request. I could also try that.
Does battery.py work every time, or only on the first try?

miikeyjay, 2024-01-10:
yes, it is working every time, and the panel is working too

[11 more replies]

closed 2025-11-15 by LennardKittner