LennardKittner/HyperHeadset #8: IOHIDDeviceSetReport failed: (0xE0005000) unknown error code

ishunyu, 2024-06-23:
Hi, thank you for this project, would really love to see it working. I was able to compile, but upon running the cli_app, I encountered issue like below. Could you see if it's something easy to troubleshoot? Thanks!
```
shun@Shuns-Air release [main] % ls
build		cli_app		cli_app.d	deps		examples	incremental
shun@Shuns-Air release [main] % ./cli_app
HidApiError { message: "IOHIDDeviceSetReport failed: (0xE0005000) unknown error code" }
shun@Shuns-Air release [main] % sudo ./cli_app
Password:
HidApiError { message: "IOHIDDeviceSetReport failed: (0xE0005000) unknown error code" }
```

ishunyu, 2024-06-23:
Some info about my headset.
```
      | +-o HyperX Cloud II Wireless@00141000  <class IOUSBHostDevice, id 0x1000330e1, registered, matched, active, busy 0 (53 ms), retain 40>
      |     {
      |       "sessionID" = 4340918369455
      |       "USBSpeed" = 1
      |       "idProduct" = 5912
      |       "iManufacturer" = 1
      |       "bDeviceClass" = 0
      |       "IOPowerManagement" = {"PowerOverrideOn"=Yes,"CapabilityFlags"=32768,"MaxPowerState"=2,"DevicePowerState"=2,"DriverPowerState"=0,"ChildrenPowerState"=2,"CurrentPowerState"=2}
      |       "bcdDevice" = 16640
      |       "bMaxPacketSize0" = 64
      |       "iProduct" = 2
      |       "iSerialNumber" = 3
      |       "bNumConfigurations" = 1
      |       "kUSBContainerID" = "11369ebf-b04d-4528-a5c9-10d935a6605a"
      |       "UsbDeviceSignature" = <510918170041303030303030303030303031000000010100010200030000>
      |       "locationID" = 1314816
      |       "bDeviceSubClass" = 0
      |       "bcdUSB" = 512
      |       "kUSBSerialNumberString" = "000000000001"
      |       "USB Product Name" = "HyperX Cloud II Wireless"
      |       "IOCFPlugInTypes" = {"9dc7b780-9ec0-11d4-a54f-000a27052861"="IOUSBHostFamily.kext/Contents/PlugIns/IOUSBLib.bundle"}
      |       "USB Address" = 5
      |       "bDeviceProtocol" = 0
      |       "USBPortType" = 1
      |       "IOServiceDEXTEntitlements" = (("com.apple.developer.driverkit.transport.usb"))
      |       "kUSBCurrentConfiguration" = 1
      |       "USB Vendor Name" = "Kingston"
      |       "Device Speed" = 1
      |       "idVendor" = 2385
      |       "kUSBProductString" = "HyperX Cloud II Wireless"
      |       "USB Serial Number" = "000000000001"
      |       "IOGeneralInterest" = "IOCommand is not serializable"
      |       "kUSBAddress" = 5
      |       "kUSBVendorString" = "Kingston"
      |     }
```

LennardKittner, 2024-06-23:
Hi, it looks like that you have the same variant of the headset as I do.
You could try applying the patch and run it again. his will at least tell me on which line the error happens.
However, even if I know where the error happens, it is verily likely that I don't know why it happens because the underlying hip error just says "unknown error code".
Could you try to running the app on a different machine to see if it works there?
[patch.txt](https://github.com/user-attachments/files/15944089/patch.txt)

ishunyu, 2024-06-23:
Thanks for taking a look! I don't have a different machine right now but I will try when I get a chance. I have a M2 machine if that makes any difference. Here's the stacktrace. No worries if it's impossible to figure out.
```
shun@Shuns-Air HyperXCloudIIWireless [main] % RUST_BACKTRACE=full ./target/release/cli_app
thread 'main' panicked at src/lib.rs:125:52:
called `Result::unwrap()` on an `Err` value: HidApiError { message: "IOHIDDeviceSetReport failed: (0xE0005000) unknown error code" }
stack backtrace:
   0:        0x102e2546c - <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt::h1f3776e0b5c7517d
   1:        0x102e3a434 - core::fmt::write::heedef092c8c0962e
   2:        0x102e236b4 - std::io::Write::write_fmt::h7178e8e2ea928914
   3:        0x102e252c4 - std::sys_common::backtrace::print::h417292deb95532ed
   4:        0x102e26390 - std::panicking::default_hook::{{closure}}::h0cb68f1228c4613a
   5:        0x102e26084 - std::panicking::default_hook::h24535936bc1f51de
   6:        0x102e26c48 - std::panicking::rust_panic_with_hook::h5db4d2345b297bed
   7:        0x102e26678 - std::panicking::begin_panic_handler::{{closure}}::h3fd558f09a0d5492
   8:        0x102e258f4 - std::sys_common::backtrace::__rust_end_short_backtrace::hfc76eebe1ce501b2
   9:        0x102e263e8 - _rust_begin_unwind
  10:        0x102e406e0 - core::panicking::panic_fmt::hc2b459a5bd3dce66
  11:        0x102e409cc - core::result::unwrap_failed::h88d7eb352f20b747
  12:        0x102e08b5c - hyper_x_cloud_ii_wireless::Device::update_battery_level::haf79d3aad9562029
  13:        0x102e07e14 - cli_app::main::h773dc93aa36cdd3d
  14:        0x102e07b20 - std::sys_common::backtrace::__rust_begin_short_backtrace::h01aa750943d0b802
  15:        0x102e07b38 - std::rt::lang_start::{{closure}}::h8651681d32da691f
  16:        0x102e211bc - std::rt::lang_start_internal::hecc68fef83c8f44d
  17:        0x102e07fd4 - _main
```