LennardKittner/HyperHeadset #9: Broken pipe on Manjaro with Cinnamon

mben42, 2024-07-01:
Hi, I get a broken pipe at start on Manjaro / Cinnamon. Is there a way to get more logs to debug that? I tried both your compiled binary and to compile myself and both CLI and tray.

LennardKittner, 2024-07-08:
Hi, I think it is very likely that you are using the older version of HyperX Cloud 2. Apparently, that version does not work very well with my app. Could you please run `lsusb` and send me the entry of your headset?

mben42, 2024-07-08:
Hi, here is the output with the headset on and plugged in:
`-> % lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 003: ID 1bcf:0215 Sunplus Innovation Technology Inc. Aukey-PC-LM1E Camera
Bus 001 Device 004: ID 1e71:3008 NZXT NZXT KrakenZ Device
Bus 001 Device 005: ID 1462:7c56 Micro Star International MYSTIC LIGHT 
Bus 001 Device 006: ID 0951:1718 Kingston Technology HyperX Cloud II Wireless
Bus 001 Device 007: ID 0951:1719 Kingston Technology HyperX Cloud II Wireless
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 002: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 003 Device 003: ID 045e:02e6 Microsoft Corp. Xbox Wireless Adapter for Windows
Bus 003 Device 004: ID 2357:0604 TP-Link TP-Link UB500 Adapter
Bus 003 Device 005: ID 046d:c547 Logitech, Inc. USB Receiver
Bus 003 Device 006: ID 046d:c547 Logitech, Inc. USB Receiver
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 004 Device 002: ID 05e3:0612 Genesys Logic, Inc. Hub
`

I bought it in March 2021, if it can help you.

LennardKittner, 2024-07-08:
Ah, yes, that is the older version of the Headset Sady. The app has some issues with that version.https://github.com/LennardKittner/HyperXCloudIIWireless/issues/1

mben42, 2024-07-21:
I close this issue as a duplicate, thanks for your help.

closed 2025-12-30 by LennardKittner