LennardKittner/HyperHeadset #10: Broken Pipe on Nobara 40 (Fedora 40)

Epzillon, 2025-03-06:
Hello! Thank you very much for making this project. I have been running Linux for well over a year without knowing the battery level of my headset.

I am using a pair of HyperX Cloud II Wireless on Nobara 40 (based off Fedora 40).

When running the `cli_app` and the `hyper_x_cloud_ii_wireless` executables I get met with the appropriate messages once the headset is plugged out and/or turned off. However when I turn my headset on it only logs "Broken pipe".

I glanced over the code quickly and found that this seems to just be a default error that gets logged since all of the other DeviceErrors are not matched.

When running `lsusb` my device is listed as follows:
```
Bus 003 Device 009: ID 0951:1718 Kingston Technology HyperX Cloud II Wireless
```

I am unsure what the issue would be here, I would assume it's some form of dependency issue as I am running Fedora, but the rest of the application seems to be running fine.

Thankful for any help or discussion revolving this!

LennardKittner, 2025-03-06:
Hi, I'm happy to hear that you like the app. Unfortunately, you have the older version of the headset (lsubs shows "Kingston Technology" instead of "HP, Inc") and there are some issues with that #1 version.
I have the newer version, so I can't easily fix the issue, and I haven't had the time to reverse engineer the windows hyperx app yet.

LennardKittner, 2025-03-15:
Hi, I started reverse engineering the HyperX Windows app, and I found some interesting details, including a battery level packet for the Cloud2.
However, someone already tried that packet on the old Cloud2, and it seems to only works once.
In my case, the packet does not work at all, but I was able to determine the packets my headset uses by attaching [dnspy](https://github.com/dnSpy/dnSpy) to NGenuity2Helper.exe and setting a breakpoint at the function that sends packets to the headset.
Would you be interested in trying that on your headset?
If so, I can provide instructions on how to attach dnspy to the app.

Epzillon, 2025-03-15:
Absolutely! Im sorry I didn't get around to replying to your first response, been a couple of hectic weeks. But yeah, just send over instructions all I'll try to fix it!

LennardKittner, 2025-03-15:
I'm happy to hear that. OK, please do the following:
- Boot Windows
- Download [ngenuity](https://hyperx.com/pages/ngenuity) and set it up
- Download [dnspy](https://github.com/dnSpy/dnSpy)
- Get Access to NGenuity2Helper.exe
  - go to C:\Program Files\WindowsApps
  - Right-click on the WindowsApps folder and select Properties
  - Go to the Security tab and click Advanced
  - Click Change next to the Owner section
  - Enter your username (or Administrators group) and click OK
  - Click Apply and OK
- Turn off your Headset
- Re open ngenuity
- Now Open dnspy 
- go to debug -> Attach to process... -> select NGenuity2Helper.exe -> Attach 
- go to File -> Open -> then open C:\Program Files\WindowsApps\33C30B79.HyperXNGenuity_5.28.1.0_x64__0a78dr3hq0pvt\Assets\Native/NGenuity2Helper.exe
- in the left section select NGenuity2Helper -> NGenuity2.Devices -> HidDevice -> `SetOutputReport(byte []) : bool`
- right click the first line in the function and click Add Breakpoint
- Next turn on your headset and open the ngenuity headset tab
- Now every time NGenuity sends a messages to your headset the breakpoint should trigger
- Every time the breakpoint is hit copy the contents of `buffer` and then click continue
- Send me the buffer contents

LennardKittner, 2025-03-15:
Alternatively, we can also do this together via Discord or TeamSpeak.

Epzillon, 2025-03-15:
I will have a quick look and see if I can get anything on an hour, otherwise I will have to take a look at this tonight or tomorrow.

Epzillon, 2025-03-15:
Ok, so to start of, the version of Ngenuity I am on is currently 5.28.0.0, do note if I need to update. I assume this since the following is in the path under WindowsApps:
`33C30B79.HyperXNGenuity_5.28.0.0_x64__0a78dr3hq0pvt`

By "the first line in the function", I would assume you mean the buffer array as follows:

[image]

I am a bit unsure of how you wanted the data structured but the following is the entire output, including the expanded buffer array. 
```
+		this	{NGenuity2.Devices.HidDevice}	NGenuity2.Devices.HidDevice
-		buffer	{byte[0x0000003E]}	byte[]
		[0]	0x06	byte
		[1]	0x00	byte
		[2]	0x02	byte
		[3]	0x00	byte
		[4]	0x9A	byte
		[5]	0x00	byte
		[6]	0x00	byte
		[7]	0x68	byte
		[8]	0x4A	byte
		[9]	0x8E	byte
		[10]	0x0A	byte
		[11]	0x00	byte
		[12]	0x00	byte
		[13]	0x00	byte
		[14]	0xBB	byte
		[15]	0x01	byte
		[16]	0x00	byte
		[17]	0x00	byte
		[18]	0x00	byte
		[19]	0x00	byte
		[20]	0x00	byte
		[21]	0x00	byte
		[22]	0x00	byte
		[23]	0x00	byte
		[24]	0x00	byte
		[25]	0x00	byte
		[26]	0x00	byte
		[27]	0x00	byte
		[28]	0x00	byte
		[29]	0x00	byte
		[30]	0x00	byte
		[31]	0x00	byte
		[32]	0x00	byte
		[33]	0x00	byte
		[34]	0x00	byte
		[35]	0x00	byte
		[36]	0x00	byte
		[37]	0x00	byte
		[38]	0x00	byte
		[39]	0x00	byte
		[40]	0x00	byte
		[41]	0x00	byte
		[42]	0x00	byte
		[43]	0x00	byte
		[44]	0x00	byte
		[45]	0x00	byte
		[46]	0x00	byte
		[47]	0x00	byte
		[48]	0x00	byte
		[49]	0x00	byte
		[50]	0x00	byte
		[51]	0x00	byte
		[52]	0x00	byte
		[53]	0x00	byte
		[54]	0x00	byte
		[55]	0x00	byte
		[56]	0x00	byte
		[57]	0x00	byte
		[58]	0x00	byte
		[59]	0x00	byte
		[60]	0x00	byte
		[61]	0x00	byte
		array	Cannot obtain value of the local variable or argument because it is not available at this instruction pointer, possibly because it has been optimized away.	
		ex	Cannot obtain value of the local variable or argument because it is not available at this instruction pointer, possibly because it has been optimized away.	
```

LennardKittner, 2025-03-15:
The structure is fine; could you please send me more packets? And maybe include the stack trace?
You can show the stack trace by going to debug -> windows -> call stack

Epzillon, 2025-03-15:
I will have to look into this more tonight, since I will be out for a few hours. I tried a few times and was waiting for a while to see more packets but, whatever was logged in the console only seemed to be output I sent above.

ataalik, 2025-03-16:
I don't know if it will make a difference but the dnspy @LennardKittner linked is discontinued and really old. [This](https://github.com/dnSpyEx/dnSpy) is an active and good fork. 

as far as your device goes in the source code I see this: 

```C#
namespace NGenuity2.Devices.Headset
{
	// Token: 0x020007D3 RID: 2003
	public class CMDHSCloudIIWirelessGetBatteryInfo : CMDHSCloudIIWirelessBase
	{
		// Token: 0x06002CEC RID: 11500 RVA: 0x000B6F8C File Offset: 0x000B518C
		public override void Execute(HyperXDevice device)
		{
			byte[] array = device.CreateBuffer();
			array[0] = 6;
			array[2] = 2;
			array[4] = 154;
			array[7] = 104;
			array[8] = 74;
			array[9] = 142;
			array[10] = 10;
			array[14] = 187;
			array[15] = 2;
			device.SetOutputReport(array);
			Thread.Sleep(50);
			device.GetInputReport(0);
		}
	}
}

```

If it works the same as cloudII wireless core when you send these bytes you should get a response from the device with the battery level.

ataalik, 2025-03-16:
The device I have will also send battery level when it goes down.

LennardKittner, 2025-03-16:
Thanks, I will check it out.
I actually tried this packet before, and it seemed to only work once and after that cause a broken pipe error. I also found another function that is supposed to set up the base packet for the original cloud 2, but it just creates a 62Byte packet with zeros. So I was interested to see what is actually sent to the device and in which order.

LennardKittner, 2025-03-27:
@Epzillon I’ve updated the project with the packet data you provided. I haven’t updated the release version yet, so if you have time, could you compile the latest commit and check if it now works with your headset?

LennardKittner, 2025-03-27:
@ataalik  The new version should also make it a lot easier to add new headsets. If you're interested, feel free to open a pull request to add support for the CloudII Core.

kmichalski0, 2025-03-30:
I apologies for out of subject comment, but I wanted to say that I would love to use your stuff as I own Cloud 2 wireless, but unfortunately it's an older version (`Bus 005 Device 011: ID 0951:1718 Kingston Technology HyperX Cloud II Wireless`) which means I can't use it :( 

So I am waiting for the new release with anticipation!

LennardKittner, 2025-03-30:
Hi, I'm happy to hear that. The main branch has already been updated and may work on your headset. You can either clone the repo and compile it yourself, or if you prefer, I can also send you a precompiled version. I don't want to release a new precompiled version yet because the headsets crashes sometimes, and I want to figure out why first.

kmichalski0, 2025-03-31:
If that's not a bother, could you send me the precompiled version, please? I'll be on a lookout for the crashes and can report back with the logs, if that's helpful to you

LennardKittner, 2025-03-31:
OK, here are the two programs. Thanks for your help.
[hyper_headset.zip](https://github.com/user-attachments/files/19535038/hyper_headset.zip)

kmichalski0, 2025-03-31:
I've tried the precompiled version but I'm getting the 'No device found'.

[image]

When I run the tray version I get this:

[image]

I have the `99-HyperHeadset.rules` created in `/etc/udev/rules.d/` and I replugged the dongle. The headset is also on.

If you want me to, I can create a new ticket, as this is for Nobara and I'm on EndeavourOS

LennardKittner, 2025-03-31:
Sorry, the previous version used the wrong vendor ID. This version should now use the correct vendor ID.
[hyper_headset.zip](https://github.com/user-attachments/files/19535460/hyper_headset.zip)

This issue doesn't really have anything to do with Nobara either and is solely due to the headset version

kmichalski0, 2025-03-31:
Gotcha! 
With this version I get a different error(s) (progress!) 
That's with the headset on and off
[image]

kmichalski0, 2025-03-31:
Hmm... interesting. I cannot remember when I bought my Cloud 2 Wireless. I think it was maybe 4 years ago? I wonder how different the packets each of our versions use. 

I don't really know how to acquire the packets you need. I could hop on Windows and try to get you some in that case. May not be today, but I'll try to do that in the following days

LennardKittner, 2025-04-02:
Sorry for the late response, and thanks for offering to help. You could try [this](https://github.com/LennardKittner/HyperHeadset/issues/10#issuecomment-2726573205)

tchofy, 2025-07-25:
I've found this repo, that works on my old cloud 2
https://github.com/Agustin-Jerusalinsky/hyperx-cloud-II-battery

It only monitors battery level, however.

LennardKittner, 2025-07-30:
@MiniBossDK thanks, I will look into it.

LennardKittner, 2025-08-02:
Hi, I have added a litter [tester](https://github.com/LennardKittner/HyperHeadset/blob/old_cloud2/src/bin/packet_tester.rs) that sends the packets you provided and then prints the response.
You can compile it yourself `cargo run --bin packet_tester`  or use the attached binary.

[packet_tester.zip](https://github.com/user-attachments/files/21559726/packet_tester.zip)

MiniBossDK, 2025-08-04:
Ran the binary you provided in Arch Linux and got a lot of Broken Pipe errors back and no responses most of the time. However, the first time I ran it I did get some response in the start:
[HyperX packets responses 1](https://pastebin.com/ADKkbWVm)

All the times after that I did not get any response. I tried to put my wireless adapter to another USB and got a single response:
[HyperX packets responses 2](https://pastebin.com/KcSeV2g2)

I doubled checked your instructions from the README and they are all correct.

LennardKittner, 2025-08-10:
Hm, very strange. I am using the [hidapi](https://crates.io/crates/hidapi) library to send packets to the headset, but apparently something is not working correctly.
Does the [Python script](https://github.com/Agustin-Jerusalinsky/hyperx-cloud-II-battery) mentioned earlier reliably work?
If so I see three possible options:
1. Updating the HID library - I don't think this will change anything.
2. Comparing the packets sent by my app with the ones sent by the Python script and the HyperX app but on the packet level with Wireshark. - This would be the best approach, but it's time-consuming without the headset. Furthermore, if it turns out that hidapi is at fault, I'd need to patch it, which is probably not easy.
3. Using the Python script or a reimplementation to talk to the headset - most likely the simplest solution

MiniBossDK, 2025-08-11:
Well, at first the python script did not work for me, until I realized today that I had to add myself to the "plugdev" group which were not mentioned in his README. It now works reliably.

LennardKittner, 2025-08-29:
Ups, yeah, writing flawless code on the first try is hard, and I can't really test it beforehand.
This should fix the error.
[hyper_headset.zip](https://github.com/user-attachments/files/22040386/hyper_headset.zip)

MiniBossDK, 2025-08-30:
Unfortunately, still the same error but a little different for both of the executables:
```
Connecting to HyperX Cloud II Wireless
Received packet: [11, 0, 187, 3, 0, 0, 0, 0, 0, 0, 0, 0]

thread 'main' panicked at src/devices/cloud_ii_wireless.rs:157:42:
index out of bounds: the len is 12 but the index is 12
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

MiniBossDK, 2025-08-30:
Running **hyper_headset** gave the following output:
[HyperX packets | HyperHeadset](https://pastebin.com/GHZB86xk)

Running **hyper_headset_cli** gave the following output: 
[HyperX packets | HyperHeadsetCLI](https://pastebin.com/eDbnBZ6b)

You may see some odd numbers in the hyper_headset_cli and that is because my headset today has been acting weirdly by not wanting to turn on without it charging. So everything here is done while it is just charging. If I pull out the charging cable it turns off immediately. It basically turned from wireless to wired now and I don't know why. :P

LennardKittner, 2025-08-31:
Thank you for your help.
It's good to hear that the output matches both NGENUITY and the Python script.
Hopefully, your headset is just acting weird and not actually broken.

The CLI app should not mute or unmute your headset unless you tell it to, so something might be off.

I also noticed there are still some broken pipe errors in the logs; it would be interesting to see what causes them.

MiniBossDK, 2025-09-09:
Alright, here is a lot of pastebins which still contains Broken Pipes and it still unmutes my headset. I have made notes in there with a <---------------- to indicate where specifically it did it.  And also written before every output what my headset configuration were.

Here I test **hyper_headset** in different configurations:
[Hyperheadset | Different configurations](https://pastebin.com/30d69sHC)

Here I test **hyper_headset_cli** in different configurations:
[HyperheadsetCLI | Different configurations](https://pastebin.com/Te9aqfNw)

Here I test **hyper_headset_cli** with sidetone and automatic shutdown:
[HyperheadsetCLI | Flags](https://pastebin.com/4LjT7sS1)

MiniBossDK, 2025-09-21:
The good news is that I just replaced my battery in my headset yesterday (pain!). And the charging and battery level is showing correctly in NGENUITY now.  

Hyperheadset got the battery level correct. But it it got everything else wrong like it says it is not charging but it is:
[Hyperheadset 2](https://pastebin.com/vEKyDQCK)

The same is true for HyperheadsetCLI:
[HyperheadsetCLI 2](https://pastebin.com/S59yfxV9)

I tried with different commands like --mute, --automatic_shutdown and --enable_side_tone but they did not do anything. The good thing is that is does not try to unmute my mic when running both of the binaries.

This is what NGENUITY shows, which is correct:
[image]

LennardKittner, 2025-09-21:
Thank you!
Could you please try it again with all settings toggled so:
- mic unmuted
- not charging
- automatic shutdown != 0 set using NGENUITY
- mic monitoring off
- surround sound off

LennardKittner, 2025-09-22:
Thank you. I think it should now detect:
- Charging state
-  automatic shutdown
-  whether it is muted
-  battery level

I am still unsure about mic monitoring and surround sound, but I added a new packet to handle them. 

The only things you should be able to change using the CLI app are:
- side tone (mic monitoring)
-  atomatic shutdown

[hyper_headset.zip](https://github.com/user-attachments/files/22473455/hyper_headset.zip)

LennardKittner, 2025-09-25:
Hi, I forgot to do a "get_input_report" before requesting the battery level in the passive refresh, which causes the broken pipe error. It is fixed now and should hopefully work. I will read the rest of your messages later, but thank you already for your support.

dinasty02091994, 2025-09-25:
> Hi, I forgot to do a "get_input_report" before requesting the battery level in the passive refresh, which causes the broken pipe error. It is fixed now and should hopefully work. I will read the rest of your messages later, but thank you already for your support.

The broken pipe error seem to be gone now, i see however that unfortunately they tray still does not work: 

[image]

I can also see that now only the battery level is reported after the inital other things, sidetone, surround, automatic shutdown etc. 

Is there any way i can be of help here to troubleshoot? not really sure where to start in terms of the tray.

LennardKittner, 2025-09-25:
Hi everyone, I think I figured out why the tray application doesn't work. My headset has a special packet to indicate if it is connected, but I haven't found an equivalent packet for your headset. Thus the app always assumes that the headset is not connected. From now on the app just assumes that the old Cloud 2 is always connected. I would be interested to see, however, what the dongle responds if the headset is turned off if someone wants to test that.

I also forgot to do a "get_input_report" before sending the packets that change stuff in the CLI app, so please try again to use `--automatic_shutdown` and `--enable_side_tone`.

@Epzillon you can use `&>` instead of `>` for logging. It will then also log stderr, and you don't have to write down any errors manually. Your headset turning off after running the application is strange. I would appreciate it if you could pinpoint specific packets that cause anomalies. However, I think I know which packet unmutes your headset.

Surround sound may be correctly detected now.

I intend to add another sequence of requests but haven't had the time yet.

Here is also a prebuilt version:

[hyper_headset.zip](https://github.com/user-attachments/files/22545334/hyper_headset.zip)

dinasty02091994, 2025-09-25:
I may have found something interesting here, i noticed building the latest version that the tray still did not work, if i run both of the applications at the same time however it does work, meaning if i start hyper_headset first and then directly after hyper_headset_cli the tray app will produce information. Running hyper_headset alone will still give the not connected error. 
Once running hyper_headset_cli once hyper_headset will continue to produce up to date information to the tray app. 

[image]

LennardKittner, 2025-09-30:
Hi everyone, I added a new sequence of packets, which is sent periodically, and hopefully fixed the tray application.

As always, there is a prebuilt version below, but you can also build it yourselfs.
[hyper_headset.zip](https://github.com/user-attachments/files/22624499/hyper_headset.zip)

LennardKittner, 2025-10-04:
Good to hear. It takes a couple of seconds because there is still a 2s debug pause between packets. It would be very helpful if you could provide me with the console output of the tray and CLI app and the headset configuration at the time of execution. So I can maybe figure out what responses the headset sends if, e.g., the mic gets muted.

You could also try again to use `--automatic_shutdown` and `--enable_side_tone`

fsvm88, 2025-10-19:
Hi, another one with the old headset :)

## General
I just tested your latest changes:
1. Apps don't crash anymore
2. Battery reading, charging status seem to work
3. Mic connected status only reports "Unknown"

## Mute
- Mute always reports `false`, even when it's enabled via the headset
- Not passing any params, the headset gets unmuted no matter what
- Passing `--mute` with false or true, the result is always unmuted headset

## Surround sound
- Always reported as `enabled` regardless of the headset actual state
- Passing no flag/false/true changes nothing, no effect can be heard (when toggling on/off via the power button there's a short audio interruption - volume fades out/in)

## Side tone
- Always reported as `enabled` regardless of the headset actual state
- Whether I pass no flag/false/true, I can hear the "side tone enabled" double-beep in the headset, but when the CLI app exits it is not enabled

fsvm88, 2025-10-19:
@LennardKittner you should have received a notification about a PR =)

I managed to export the entire NGenuity decompiled project via the updated dnSpy, and asked Copilot (Claude Sonnet 4.5) to document the protocol *for all headsets*.

Then I made it implement the protocol for my headset. It was allowed hardware access to validate all changes.

During the implementation a couple things turned out:
- some commands are implemented only via Windows APO (e.g. Surround State can be read, but not written, because that'd require Windows APO)
- some things are even hard read-only (mute state)
- there's an unknown command 4 that is periodically reported by the headset, but according to Copilot's analysis of NGenuity, this is also *not* handled in NGenuity itself

I made it write out the full headset protocol for all headsets, and it decided on its own to write out a FEATURE_CAPABILITIES.md file to keep track of what worked and how.

Idk why it decided that `--surround_sound` and a couple other flags were unrequired and removed them, I'll try to have it restore them later :) EDIT: Ah, perhaps this branch was forked off a very old commit, and those flags were not implemented.

fsvm88, 2025-10-19:
No worries, I understand, I always review all of the generated code and normally ask the agent to falsify it if possible. Yes, things work properly now :)

I asked it to stay close to your coding style without rewriting too much, so that it doesn't look like entirely foreign code to you - I think it did a decent job except for one the big command tuple-match that it rebuilt into multiple matches.

Feel free to take the PR apart or pluck out the stuff you want, it's totally fine from my side 👍

closed 2025-11-15 by LennardKittner