LennardKittner/HyperHeadset #11: Adding support for Cloud II Core Wireless

ataalik, 2025-04-03:
Hello @LennardKittner  I am looking through the new code. it seems you have added support for set and get packets but the version I have also has the ability to send unsolicited packets, like for example when the charging cable is plugged in. Do you not have those or do you think parsing them are not needed?

LennardKittner, 2025-04-03:
Hi, I'm happy to see that you are interested in adding the Cloud II Core.
Since we can query all relevant information I didn't think this was necessary, but we can add it back by periodically calling `wait_for_updates`. The parsing can be done inside `get_event_from_device_response`.
This may actually be a good idea because I think querying the state too frequently can lead to instability.

ataalik, 2025-04-03:
Yes exactly, you only need to query once you connect to the device. The rest of the time it sends the events on it's own. I am using your old code with blocking wait_for_updates that times out and everything seems to be working fine. I also wrote a non blocking async version as an exercise but I guess it does not really have any benefits since you have to add a bunch of sleeps anyways, it becomes the same thing.

LennardKittner, 2026-02-25:
I will close this issue since Cloud II Core is now supported.